Entwicklungsburo-Stresing/EBST_CAM #94: Live view is missing out the last scan with software polling

stresingbuero, 2023-06-20:
nos = 20
camcnt = 2
nob = 2
software polling = 1

this is sus
```
scan counter 41, Nospb 10, camcnt 2
block 2, scan 0, samples_done_all_cams 40, samples_done_per_cam 20, count_of_scans_of_completed_blocks 20
```

whole log:
```
scan counter 40, Nospb 10, camcnt 2
block 1, scan 9, samples_done_all_cams 39, samples_done_per_cam 19, count_of_scans_of_completed_blocks 10
Scan: 39
Data to copy: 2176
DMA buffer read position: 0000024C60704B80
User buffer write position: 0000024C621E6B30
Reset priority
scan counter 41, Nospb 10, camcnt 2
block 2, scan 0, samples_done_all_cams 40, samples_done_per_cam 20, count_of_scans_of_completed_blocks 20
Scan: 40
Data to copy: 0
Wait for all interrupts
All interrupts done
Stop S Timer
measurement count: 3
Reset hardware counter
reset userBufferWritePos to 0x0000024C621D1FB0
Start copy data to user buffer thread.
Set priority to 31
Poll DMA buffer to user buffer started. drvno: 1
DMA buffer address: 0000024C606F0000
DMA buffer end: 0000024C60903400
Data to copy in bytes: 87040
Increase hardware block counter
Size of one scan in bytes: 2176
Address of user buffer: 0000024C621D1FB0
Wait for block trigger

get DSC 1
get DSC 2
Calc Trms failed. lastSample must be greater than firstSample and both in boundaries of nos, drvno: 0, firstSample: 9, lastSample: 999, TRMS_pixel: 0, CAMpos: 0, Nospb: 10
scan counter 0, Nospb 10, camcnt 2
block 0, scan 0, samples_done_all_cams -1, samples_done_per_cam 0, count_of_scans_of_completed_blocks 0
```

stresingbuero, 2023-08-22:
Not sure if this is the problem, but the first scan always starts as -1

Log for the first scan:
```
scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

scan counter 0, Nospb 10, camcnt 1
block 0, scan -1, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0
```

Possible error mentioned:
block 0, **scan -1**, samples_done_all_cams -1, samples_done_per_cam -1, count_of_scans_of_completed_blocks 0

sinneD-O, 2023-09-22:
d33d415d0ae6b56bf73d571354a41043d3b215c4

closed 2023-09-22 by sinneD-O